NebraLtd/Helium-Guides #314: frozen at location verification screen

scout1966, 2022-02-13:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

shawaj, 2022-02-13:
This sounds like an issue with the Helium App. You will need to report that at their repo

closed 2022-02-13 by shawaj

scout1966, 2022-02-13:
forgot to add tried, with both direct ethernet connection, and wifi connection, to  different network router, one a cell based router, and the second router linksys, connected through a LowRan type of service.