NebraLtd/Helium-Guides #317: rockpi block diagram

shawaj, 2022-02-27:
FYI @salmanfarisvp - https://github.com/NebraLtd/Helium-Guides/blob/main/docs/indoor-rockpi-hotspot/overview.md#block-diagram
[image]
[image]

salmanfarisvp, 2022-02-28:
Thanks for sharing, Added https://github.com/NebraLtd/Helium-Guides/commit/15c2e251ffc022bbb81137dacc2eee415e5949dd

salmanfarisvp, 2022-03-01:
@shawaj Re-designed ROCK Pi block diagram, what do you think? 

[image]

salmanfarisvp, 2022-03-04:
Thanks @shawaj , updated for ROCK Pi https://github.com/NebraLtd/Helium-Guides/commit/ed7ab6ec6267f0cc22899d99eee305602a8d3c05 and Indoor and Outdoor too https://github.com/NebraLtd/Helium-Guides/commit/e49f580a85d16a31c9f9bcd0fcfaf894d8d5e803

shawaj, 2022-03-04:
Nice

ChristopherRush, 2022-03-07:
@salmanfarisvp Can I use this for the datasheet also?

salmanfarisvp, 2022-03-07:
@ChristopherRush Sure, I can share the original file if you need it.

ChristopherRush, 2022-03-07:
@salmanfarisvp Yes, that would be great. Thanks.

salmanfarisvp, 2022-03-07:
Sure, here it is. All diagrams are in a single file, you can select and export selected once. Let me know if you need any help on this 🙂 [nebra-hotspot-block-diagram.zip](https://github.com/NebraLtd/Helium-Guides/files/8197108/nebra-hotspot-block-diagram.zip)

closed 2022-03-07 by salmanfarisvp