NebraLtd/Helium-Guides #322: reduce image and pdf size

shawaj, 2022-02-28:
we should try to reduce PDF size by trimming the size of images (this would also help with load times online).

This should be doable regularly using some github action

shawaj, 2022-02-28:
For example 

- https://github.com/emareg/compressPDF
- https://github.com/amaurypm/reducePDF
- https://github.com/petems/image-resizer-gh-action

salmanfarisvp, 2022-03-03:
Since this repo is open and we can use  Imgbot, it is a friendly robot that optimizes your images and saves you time. Optimized images mean smaller file sizes without sacrificing quality. https://github.com/marketplace/imgbot/plan/MLP_kgDNGw8#pricing-and-setup

shawaj, 2022-03-04:
Yeah that looks good and looks like it would be free in open source repo

shawaj, 2022-03-07:
Installed that plugin now 👌

shawaj, 2022-03-09:
annoyingly - the plugin does not seem to work.