NebraLtd/Helium-Guides #329: Create guides for setting up RAK and other devices on Balena OpenFleets

shawaj, 2022-03-30:
We now have various balena OpenFleets on balenaHub:

- RAK - https://hub.balena.io/nebraltd/helium-rak
- Helium OG - https://hub.balena.io/nebraltd/helium-og
- Sensecap - https://hub.balena.io/nebraltd/helium-sensecap
- Finestra - https://hub.balena.io/nebraltd/helium-finestra
- Controllino - https://hub.balena.io/nebraltd/helium-controllino
- Pisces - https://hub.balena.io/nebraltd/helium-pisces 

We should add guides for how to reprogram the SD cards on these to work with our software.

And in the future how to then enable remote dashboard support on them (once available)

shawaj, 2022-05-15:
@ChristopherRush reckon you could update this to include the method for downloading the image from OpenFleets?

I know it's pretty self explanatory but some people have been clicking "fork this fleet" instead of "get started" 

Also you can use the short URLs:

https://nebra.io/rak
https://nebra.io/sensecap
https://nebra.io/pisces
https://nebra.io/finestra
https://nebra.io/heliumog
https://nebra.io/controllino

ChristopherRush, 2022-05-15:
Yes, we already do have a guide on this but it needs updating and made a
bit clearer. https://helium.nebra.com/#/handy-guides/flash-sd