NebraLtd/Helium-Guides #335: Miner issues indoor and outdoor

santaclauze, 2022-05-28:
Hi, 

I had issues with my miner to begin one.

After weeks I discovered that my port 44158 was not open. Fixed that. Now my outdoor miner is syncing and is showing the following:
[image]

 It seems to not be relayed, yet the syncing to 0.02% has taken 12 hours. 

I ordered my nebras last year in March, received them about 2 months ago and have been very patient trying to get them running.

I also have another Nebra, indoor which is displaying relayed even though on the same network:
[image]

I was told that this had a hardware issue which will require an image which was already transmitted to me. I will  be performing the update this week.

Any other things I must know or you can do on your side to get this working?

santaclauze, 2022-05-28:
My outdoor miner has just become relayed by the way. Go figure why. Do you need me to submit diagnostics?

santaclauze, 2022-05-29:
outdoor hotspot now displaying:
Diagnostics has encountered an error: HTTPSConnectionPool(host='fuzzy-marmalade-warlock.skittles.stakejoy.com', port=443): Max retries exceeded with url: /v1/blocks/height (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xf5250e80>: Failed to establish a new connection: [Errno -5] No address associated with hostname'))

santaclauze, 2022-05-29:
Now it is displaying the follwing:

<html>
<body>

ECC Detected | False
-- | --

</body>
</html>

ChristopherRush, 2022-05-30:
@santaclauze This isn't a support forum. Please open a support ticket here - https://support.nebra.com/support/home

closed 2022-05-30 by ChristopherRush